fossasia/pslab-hardware #3: Libraries not found in the .pro files

sudheesh001, 2017-04-03:
The error message received is as follows:

[image]

jithinbp, 2017-04-03:
Hi,

Sorry about that. i did not notice the shared libraries. Will upload by
tonight.
In the meantime you could try skipping the error, and it might pick up the
cached library

jithinbp, 2017-04-03:
please check the latest pull request. Added proto1.lib 
I see the same errors, but I can simply skip it, and the schematic picks up the files from proto1-cache.lib . Can you check?
[image]

wavicles, 2017-04-07:
Closed with #4

commit 0d8e64f9 (2017-04-07): Merge pull request #5 from jithinbp/master

Fixes #3 Adds missing libraries
commit df297b33 (2017-04-07): fixes #3 Adds missing libraries
closed 2017-04-07 by wavicles