fossasia/pslab-hardware #6: Add info/documentation to Readme and addition docus in folder /docs and documentation in folder /docs/img/

mariobehling, 2017-04-07:
Let's follow the same structure of documentation as in our software projects. Please add additional information into Readme.md e.g. chat channel, communication, project structure, best practices and images.

Also keep documentation images for the project in /docs/img/.

A good example for documentation in a repository is here: https://github.com/fossasia/open-event-orga-server

mariobehling, 2017-04-10:
@jithinbp Please add some more info and images to the Readme.md.

mariobehling, 2017-04-10:
Please do the following
* [ ] Use international English for punctuation and spelling
* [ ] complete the parts list and provide links to parts
* [ ] correct the markup of links
* [ ] Merge section _hardware spe_ and _section Feature list for the acquisition and control_ and explain the purpose of each hardware spec/item

jithinbp, 2017-04-10:
@mariobehling I have a query -
- For the last three items on the list , are you perhaps referring to the repo 'pslab-firmware' ?
The parts list is currently in the Readme.md of pslab-firmware  . I'm moving it to this repo's readme.md

mariobehling, 2017-04-10:
Let's put things where they belong (of course sometimes there is an overlap).

Please see it from a developer perspective: "I am a young developer who sees this projects. I find it interesting. I do not know so much about hardware development. I am just starting. What do I need to build this hardware and start developing?"

Please provide all details and info here. Cheers.

closed 2017-04-17 by mariobehling

jithinbp, 2017-04-17:
> Closed #6 <https://github.com/fossasia/pslab-hardware/issues/6> via
> 2a38f1d

thanks!

commit 07ba4edb (2017-04-10): Partly Fix #6

- Partial Parts list added to Readme.md
commit 09ec7f06 (2017-04-10): Partly Fix #6

- Partial Parts list added to Readme.md
commit 48c724e9 (2017-04-10): partly fixes #6
commit 6e1c50ac (2017-04-10): Partly Fix #6

- Partial Parts list added to Readme.md