fossasia/pslab-hardware #9: Foot print components misaligned!

CloudyPadmal, 2017-05-08:
I checked the foot print of the PCB and it looks like this to me! Is it same for others?

[image]

jithinbp, 2017-05-08:
That's exactly how it looks to me. Not finished routing.
We need to make some hardware changes, so I'll have to spread out the
components and reroute.

- Add a Transistor+Zener based regulator for +/- power supplies instead of
the current resistor+zener regulator which is not efficient
- Move the I2C connector away from the CPU & give its Vcc pin some heavy
decoupling. This is because some I2C modules have tantalum capacitors for
decoupling. Ta capacitors have very low ESR and the Vcc of the chip itself
drops very quickly when they are plugged in.
- Instead of connecting Vcc of the I2C connector to 3.3V , we can consider
connecting it to 5V instead. I've found that more sensor modules have
connections like 5V GND SCL SDA instead of 3.3V GND SCL SDA even though
most work with 3.3V also since LDO regulators are used by them

- Replace 12MHz through hole crystal with 4-pin SMD crystals.

Are you well versed in KiCAD?

Also : Have the layout libraries properly been included in the git
repository? Schematic libraries had to be added later because they were
referencing a local folder on my PC.

thanks

CloudyPadmal, 2017-05-08:
Hey @jithinbp !! I haven't used KiCAD before but I went through their documentation yesterday! I've only worked with OrCAD and Proteus! :blush: 

There is a warning coming saying;
[image]

jithinbp, 2017-05-08:
I thought sudheesh had checked it after I added the library files. You might have to add the library  in the repo to KiCAD's path variable

CloudyPadmal, 2017-05-08:
Should I add it in this repo?

jithinbp, 2017-05-08:
I'm a little confused here.    Did you mean you will add the library? It's already there : proto1.lib
You can add it to the path variable, and if everything looks good, I'll pull the same

CloudyPadmal, 2017-05-08:
Ah yea! :D I added it to my kicad library path and now it's all fine :D :+1:

jithinbp, 2017-05-08:
cool! Maybe you can starting making sense of how it works.
Most of the schematic should be easy to understand. 

The NRF module is a major hassle during production. I hadn't planned on keeping it as part of this schematic, but in case we decide to use it, I think it might be easier to use the [SMD modules](https://www.aliexpress.com/item/5pcs-Free-Shipping-NRF24l01-2-4GHz-Wireless-Module-Mini-Version/32430542898.html) . Or, use the IC itself, and make the relevant circuitry including a PCB trace antenna. The 16MHz clock required for its operation can be derived from the PIC24 itself.

thoughts?

closed 2017-05-08 by jithinbp

CloudyPadmal, 2017-05-08:
Embedding the IC onto the current schematic would be better right? That module is like 2cm x 1cm but if we can implement it on the board itself with clock also derived from PIC, the space will be preserved I think! For the antenna we can even provide an IPX connector and an antenna right? :blush:

jithinbp, 2017-05-08:
The IC would be ideal, but it would be quite the cost addition. Two different versions can also be produced. Also, from what I've heard, the original IC from Nordic Semiconductors costs about $10 as opposed to the $1 chinese knock-offs that copied the silicon die itself to some extent along with the communication protocol. So we'll be going into production using knock-off chips. 
If the module is used, users can simply solder it on if they need it. Similar to the space provided for the ESP8266.
They can also choose to install a module with a built-in PA+LNA & rubber ducky antenna if they require longer range. modules are a flexible option.

CloudyPadmal, 2017-05-09:
If a space is left for user to solder a module himself, then we can simply implement the interface for the module right? Or is it a not so nice practice to provide just the female headers for connectivity of a module with male pins?
[NRF24L01](http://www.ebay.com/itm/1pcs-NRF24L01-2-4GHz-Antenna-Wireless-Transceiver-Module-NEW-C-/141975887566?hash=item210e6c1ece:g:gUQAAOSwLa9UXJmy)

That way, lot of space can be spared right?

CloudyPadmal, 2017-05-10:
Hello @jithinbp !!
Can I try to replace the TH crystal with an SMD crystal? :)

jithinbp, 2017-05-11:
Certainly. Can you try it in a separate layout file?
The crystal's schematic already exists in proto1.lib . It's a passive crystal with diagonally opposite terminals.  The other diagonal is grounded

You can also check out [Fox 924b](https://www.digikey.com/product-detail/en/fox-electronics/FOX924B-25.000/631-1074-1-ND/1024779) . It needs Vcc, GND , and outputs a highly accurate (+/-2.5ppm) 25MHz clock. Only two or three lines in the firmware will need to be changed for this hardware revision