fossasia/pslab-hardware #17: Modify schematic library proto1.lib to comply with KiCAD's library testing utility

jithinbp, 2017-06-08:
PSLab KiCAD/proto1.lib does not currently pass the automatic schematic library testing utility provided by KiCAD.
The following procedure can be followed to check the errors
```
git clone https://github.com/KiCad/kicad-library-utils 
sudo kicad-library-utils/schlib/checklib.py <path to PSLab\ KiCAD>/proto1.lib 
```

at present, violations of the following nature are shown by the checklib.py utility
```
Checking symbol 'myUSBMB':
  Violating Rule 4.10 - Part metadata
  Violating Rule 4.12 - Footprint filters
  Violating Rule 4.2 - Symbol visual style
  Violating Rule 4.5 - Pin orientation
  Violating Rule 4.9 - Default fields
Checking symbol 'tp':
  Violating Rule 4.1 - Pin placement
  Violating Rule 4.10 - Part metadata
  Violating Rule 4.12 - Footprint filters
  Violating Rule 4.8 - Field text size
  Violating Rule 4.9 - Default fields
```

jithinbp, 2017-06-08:
A pretty amazing blog post of what can be achieved using travis can be found [here](http://scottbezek.blogspot.in/2016/04/automated-kicad-openscad-rendering.html)
It explains how travisCI can be used to automatically generate and display the CAD rendering in the README doc

jithinbp, 2017-06-08:
an automatic render of the schematic can also be made. [tutorial](https://scottbezek.blogspot.in/2016/?m=0)

CloudyPadmal, 2017-06-08:
We'll have to follow a similar procedure in `KiCAD` right since it is a CAD tool? :)

jithinbp, 2017-06-08:
First goal is to get the libraries to be accepted by the automatic script, otherwise we will keep getting build errors with Travis.
I am unable to find the rule list, and how to correct them. using trial and error. e.g. , `  Violating Rule 4.1 - Pin placement` is not verbose enough. The schematic works fine with KiCAD, but is somehow not accepted by the checking utility. Please share the rule list if you find one. I have found a few Travis logs for failed builds from other projects, and am locating sources for the builds that passed , so that I can compare the same with our library.

The blog posts I had linked to can be stretch goals. It uses Travis to generate and upload the latest render to a cloud link, and the same is embedded in the README.md

CloudyPadmal, 2017-06-09:
Sure! I'll share if a rule list is found! :+1:

CloudyPadmal, 2018-02-08:
Travis build is now passing with `library-check.sh` of KiCAD utilities repo, this issue can be closed!

closed 2018-02-08 by CloudyPadmal