fossasia/pslab-hardware #38: Sell ready-made PCBs?

shakram02, 2018-04-09:
Hello,

I'm interested in having PSLab hardware. But I don't know the steps to fabricate one, since SMT assembly isn't available in my country.

Are you willing to sell ready-made PSLab kits through your website?

If no,
I've found [PCB cart](https://www.pcbcart.com/assembly) to make some copies. I don't know which format they accept.

It would be nice if you package the files in an amateur-friendly `zip` file with the instructions and your preferred PCB manufacturer if possible

Thank you

CloudyPadmal, 2018-04-14:
Hello @shakram02 ! The gerber files to manufacture the PCBs can be found [here](https://github.com/fossasia/pslab-hardware/blob/master/PSLab%20KiCAD%20v4/Gerber_For_PSLab.zip) and the bill of materials also can be found in [here](https://github.com/fossasia/pslab-hardware/blob/master/PSLab%20KiCAD%20v4/PSLab_bom_for_Mouser.txt)
For manufacturing you can go with any manufacturer you prefer!
Thanks!

shakram02, 2018-04-14:
Thank you @CloudyPadmal,

I recommend that you have an online store or sell it via a vendor, as you know PCB manufacturers only accept batch orders. I'm a student and won't be able to pay for 10 pcs to use only one. You can also get profit if you sell it online.

Thank you :+1:

shakram02, 2018-04-17:
look :sob: 
This is the cost from a manufacturer. I uploaded the files you mentioned. I need to buy  and 10PCS so the total is $908

[image]

CloudyPadmal, 2018-04-17:
Yes @shakram02 , I get the point! :disappointed: The quotation must be for a 2 layer board but PSLab is a 4 layer board; making the $5.59 upto $50 may be. Let's see if we can get you a device! cc : @mariobehling

shakram02, 2018-04-18:
@CloudyPadmal @mariobehling I'd love PSLab to be available for everyone also and not just me to get a device, maybe you can contact spark fun/ adafruit to sell them for you and get you a profit margin?

The reason is; In my university we have measurement devices from the 70s, and each 10 students stand around one device. That's extremely miserable and not fun at all, by making PSLab streamlined anyone can buy the kit easily and for the suitable price, while making you some profit.

Thank you :+1: