fossasia/pslab-hardware #49: Considerations for Hardware Production Version 2018

mariobehling, 2018-08-12:
Below are considerations for hardware production from email exchanges with our partner at the Fraunhofer Institute and PCB manufacturer.

> The BOM allows to extract these information as well in manual fashion, however, most CAD tools will provide this info alongside the above mentioned pick and place information in one machine-readable file. So, if you verify the capabilities of KiCAD on the former, please check also on the latter requriements.
If KiCad is not able to do so, well, then back to manual :), same goes with the keep-out areas underneath the PCB for positioning of support pins during the printing process. With a "nest" (see below), this will not be needed.

> As to the PCB itself - could you check with your PCB supplier whether he could put the PCBs in a manufacturing panel ready for singulation either by milling, rolling blade separator (e.g. https://www.cab.de/de/elektronik/nutzentrenner/maestro-2-2m/) or hook knife (e.g. https://www.cab.de/de/elektronik/nutzentrenner/stegtrenner-hektor-2/). Milling will get the best contour precision (this is what you have now, but with separation prior assembly), the others will leave some minor jagged edges.
If the vendor is happy to panelize this for you, I would suggest a minimum 2x2, but for the sake for flexibility and speed preferrably a 3x6 (or any other option up to 460x460mm, suitable for print, assembly and soldering).
Here now the additional request - for the panels, we would need alignment marks, circles with min. 1mm/max 3mm diameter, exposed from any solder resist, to have the cameras identify the "copper pad positions" and then place the components into their final position. Without any alignment marks, the assembly system is "blind". This -in consequence- means if say we need to work on the singulated PCB, not only the total assembly time will go up, but also a dedicated nest with alignment marks would need to be built, the singulated PCB positioned relatively precisely into that nest, thus making the PCB ready for full-auto assembly. Can also be done, but is less elegant and "manufacturing like", compared to a panelization service by the PCB vendor :)

> For panelization, please note in the discussion with the vendor these additional needs
> - free area on the two "long" sides to allow for PCB-panel clamping: min3, preferred 5mm each side
> - preferred surface finish is NiPdAu ("enepig"), NiAu, tin (Sn) or Ag (most expensive!!). Also OSP can be used BUT note that any lower side population or repair will then be more critical due to the oxidation.
Please ask you vendor for a sample PCB to assess the solderability with the paste/reflow combination.

> Now to an additional (future adressing) aspects.
It seems that the core elements could be "squeezed" into modules, which can be modified if a new generation of µC or DAC appears. This will forego the need to design the whole PCB anew for the new generation, as it all can be handled on the module. Also, these modules can be significantly shrunk, if miniaturization needs arises in the future. You may want to consider this thought in a design revision of the future.

mariobehling, 2018-08-12:
PCB manufacturer feedback:

1. Output of coordinates of component central location
KiCAD generates pos files with centroid informations. The file content will look something like this;
2. BoM with both part details and assembly information
KiCAD doesn’t support a combined feature. We’ll have to do this manually.
3. Keepout areas underneath the PCB for positioning of support pins
“Same goes with the keep-out areas underneath the PCB for positioning of support pins during the printing process. With a "nest" (see below), this will not be needed.”
Design doesn’t have any components on the bottom side. So do we need any keep-out area for placement? I’m a little unclear about the “nest” as there was nothing attached below.
4. PCBs in a manufacturing panel
I checked with PCBWay and they provide a panelizing feature. We can make a panel and submit or ask them to create a panel for us.
They provide V-Scoring and/or Tab-Routing as the routing to separate boards.
5. Alignment marks on PCBs
Fiducial marks can be added to the existing design. I suppose a similar implementation like in the figure will do it, the alignments :)

6. Panelization specifications
- Free area on the two "long" sides to allow for PCB-panel clamping: min3, preferred 5mm each side
Possible
- Preferred surface finish is NiPdAu ("enepig"), NiAu, tin (Sn) or Ag (most expensive!!). Also OSP can be used BUT note that any lower side population or repair will then be more critical due to the oxidation.
PCBWay offers [HASL with lead], [HASL lead free], [ENIG], [Immersion gold], [OSP] and [Hard Gold]

mariobehling, 2018-08-12:
Reply from Fraunhofer:

This is very good news!

As no double sided assembly is required, we are free to position the support pins anywhere. Thanks for having clarified this. Also, with the option to go for the panel, no milled Al-nest (i.e. a negative form to receive the PCB to run it through the assembly line) is required.

As to the depanelization – what would be your preference (keeping your final form for sales in mind) -> V-groove may be cheaper but has some more rough edges, tab-routing will call for slightly larger inter-PCB distances (as of IPC 7351), but result in much cleaner edges. If you plan to mount the boards for your final product in a housing anyway, you may want to consider the price difference to your benefit!

Also on the cost side goes the PCB finish. We strongly discourage from HASL (lead) -> will bring you in trouble when running the RoHS and WEEE compliance check. With the current component sizes you have, HASL (leadfree) will still be acceptable w.r.t. pad surface flatness – as soon as you  go towards smaller components, this will eventually lead to assembly defects. ENIG w/immersion gold is currently the standard for volume production of hi-rel goods, so this is definitely to consider. Also OSP will be okay if the OSP chemistry provided by your PCB manufacturer can withstand minimum 2 thermal processes (1x full reflow, 1x soldering of the pins for the through hole components) – usually this should not be a problem, but I suggest to ask for a positive feedback to avoid issues later on. Immersion gold over copper (if I got this info right) is not a good choice, as diffusion is fast and will make the surface critcial to solder, especially if some storage time (ie. Producing first 100, then another 100, then 200 and 500  products) will deteriorate solderability. I strongly discourage the use of hard gold (plated), as –while initial soldering looks excellent- the overall reliability will be poor, making you a bad example of planned obsolescence in less than 2 years after manufacturing J

Alignment mark position and dimensioning is very good, thanks!

So, the next steps would be to exchange on the modified design with the alignment marks, receive the panel design file (stencil will be created for this entire size), freeze all designs, start buying components and PCB and go to us for the first full auto manufacturing run!

mariobehling, 2018-08-12:
Question of Padmal:

Once we make a panel, say 3 x 2 boards in one panel. For the automated pick and place machine, should we provide component coordinates to all the components on the panel or providing coordinates to only one board is enough? (Will it offset coordinates to other components of the same ID?)

cweitat, 2018-08-12:
@woshikie do have a look at this as well.

CloudyPadmal, 2018-08-13:
Reply from Fraunhofer:

One additional point to my previous answer.... For the Stencil, the full panel info on the opening site is preferred, saves us some days for measurement, panelling of stencil coordinates and ordering...

CloudyPadmal, 2018-08-13:
Padmal:

With KiCAD we can generate positions of the same component in multiple places. If we prepare the stencil matching that I guess we can straightaway configure the pick and place machine to assemble the whole panel at once :)

[image]

PR #114 (2020-08-27): fix: upgrading hardware design
This pull request is addressing all the issues left in [Milestone 1](https://github.com/fossasia/pslab-hardware/milestone/1)

1. fix #104 : removed crosshair (drill origin map) 
[image]
2. fix #97 : increased cutoff frequency using 0.5pF capacitor replacing 3.3pF
3. fix #105 : USB type C with 12 pins
4. Added UART reset connector for ESP, Bluetooth and USB connections
5. fix #75 : External power supply support
6. UART support on dual ports. (Possible to use two UART connections simultaneously)
7. fix #106 : I2C pins inline with OLED screens (VCC/GND swapped to su [... cut]
closed 2020-08-27 by CloudyPadmal