fossasia/pslab-hardware #68: two layer version for reduced costs

shakram02, 2018-11-02:
Is it possible to make pslab available in a two-layer version? having 4 layers makes PCB fabrication pretty expensive. 

Maybe you can increase the area for it to be like a tablet if needed, in my opinion, it's more important that the circuit becomes cheap and easy to fabricate than to keep it small in size

Thanks

CloudyPadmal, 2018-11-03:
Hello @shakram02 ! It could be possible to make a two layer board with expanded area. But this will require tremendous amount of time and effort to balance loads and signal tracks and it will have some other issues as well;

1. Current design dimensions are roughly around 100mm x 50mm. If we take a 2 layer board into account assuming that we are making 5 boards, this would cost about $5 and if we increase one millimeter from width making it 101mm x 50mm, the new cost will be $30 for 5 boards. (Yes of course we can increase 50mm side but will we be able to stay inside the 100mm x 100mm area is a question)

2. Bigger the circuit, bigger the interference to signals as they are traversing a longer distance. Especially to the tracks connecting sensitive instruments such as oscilloscope circuitry.

4. When the mass production is happening, the cost difference between 2 layer and 4 layer is bearable and it will actually save some money when it comes to shipping as the weight of the board is lighter in smaller 4 layer boards.

5. It won't be "Pocket" Science Lab anymore :)

#### And I got good news. The new stocks are ready and they will be available to order online pretty soon. I will ping you here when the ordering is ready for your convenience. It will be just a few more dollars than a regular Arduino Mega board :)

shakram02, 2018-11-03:
`
And I got good news. The new stocks are ready and they will be available to order online pretty soon. I will ping you here when the ordering is ready for your convenience. It will be just a few more dollars than a regular Arduino Mega board :)
`
That's super cooool :boom: