fossasia/pslab-hardware #79: PSLab Hardware V2 - new chips

mariobehling, 2018-12-09:
There are better chips available for future versions of PSLab. One recommendation is LPC43S70FET256: 32-bit ARM Cortex-M4 + 2 x M0 MCU; 282 kB SRAM; Ethernet; two HS USBs; 80 Msps 12-bit ADC; configurable peripherals, AES engine

https://www.nxp.com/products/processors-and-microcontrollers/arm-based-processors-and-mcus/lpc-cortex-m-mcus/lpc4300-cortex-m4-m0/32-bit-arm-cortex-m4-plus-2-x-m0-mcu-282-kb-sram-ethernet-two-hs-usbs-80-msps-12-bit-adc-configurable-peripherals-aes-engine:LPC43S70FET256

Because of the on board 80 MSPS ADC, 6 channel and so the oscilloscope function would be able to capture faster signals and 12 bit accuracy should be sufficient enough for most uses.

cweitat, 2018-12-23:
Alternatively, to look at FPGA for better specifications

geekyjoyce72, 2018-12-23:
> Alternatively, to look at FPGA for better specifications

Hmmm but still, performance in terms of analog and digital sampling/signal generation is really down to the AD/DA converters and their specs. FPGA will drive the cost and design complexity up needlessly, compared to a MCU like the one suggested.

I don't see an application here that justifies the use for FPGA. Thoughts?

cweitat, 2018-12-23:
@geekyjoyce72 is just an alternative that we can KIV. :)

geekyjoyce72, 2019-03-01:
https://www.nxp.com/products/processors-and-microcontrollers/arm-based-processors-and-mcus/lpc-cortex-m-mcus/lpc4300-cortex-m4-m0/32-bit-arm-cortex-m4-plus-2-x-m0-mcu-282-kb-sram-ethernet-two-hs-usbs-80-msps-12-bit-adc-configurable-peripherals:LPC4370FET100

Actually, there's also a 100 pin version, this will make routing easier to handle. 

I honestly am thinking about building one of these boards with the LPC MCU myself first, what do you think?

mariobehling, 2019-03-01:
Nice! Please go ahead and post a basic design in a new folder in this repo.

orangecms, 2020-11-29:
I am currently looking at a handful of development boards that I recently got to see what platforms are nice to develop on.

AT SAM D MCUs and other AT SAM MCUs are probably sufficient to bring all the features we need to switch platforms.
I have a [XIAO board](https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html), which is a SAM D21. I'm trying various environments, looking into [Zephyr](https://docs.zephyrproject.org/1.13.0/reference/kconfig/choice_24.html) and [FreeRTOS](https://github.com/BriscoeTech/Arduino-FreeRTOS-SAMD21), possibly [RIOT OS](https://github.com/RIOT-OS/RIOT/tree/master/cpu/samd21), [bare Rust](https://github.com/atsamd-rs/atsamd), and [Arduino](https://github.com/arduino/ArduinoCore-samd), the latter being quite complex regarding build setup for CI.

I am not exactly sure what is necessary in order to get the hardware features that the current PSLab has, so I am mostly looking from a software/firmware development perspective.

Here is a decent overview of the MCU market: https://jaycarlson.net/microcontrollers/

fcartegnie, 2021-01-18:
As dev boards STM32 discovery kits do provide STLink on board (dual USB) for free. old STMF1 are cheap.
STM32F1 series are too limited

The chinese DSO138 Kit using STM32F1 has 1Msps over 1 chan and provides LCD screen for 15$. There's no point of making a similar device for higher price and less hardware.

The PIC "secondary" 2Msps chip can still be driven by an STM to provide 4chans. (assuming we can have enough bw between the 2)

If we need to target a single chip, then we don't care about frequency. We must focus on RAM and ADC capabilities. Having a 72MHz chip that performs worse than PIC is useless, because the ADC Clock does not run at the MCU speed.

Ex: STMF3 provides DMA and RTC. There's far more chip versions. Should consider "ADC Clock" first.

https://hsel.co.uk/2015/12/10/stm32f0-high-sample-rate-adc-usage/
https://www.stmicroelectronics.com.cn/content/ccc/resource/training/technical/product_training/group0/3c/ee/54/92/43/f1/4e/27/STM32H7-Analog-ADC_ADC/files/STM32H7-Analog-ADC_ADC.pdf/_jcr_content/translations/en.STM32H7-Analog-ADC_ADC.pdf

fcartegnie, 2021-01-18:
Provides summary table with max SPS per chip
https://www.st.com/resource/en/application_note/dm00129215-extending-the-dac-performance-of-stm32-microcontrollers-stmicroelectronics.pdf

orangecms, 2021-01-20:
[image]

for quick comparison, here's the table from the DAC performance document