fossasia/pslab-hardware #107: Move ESP to frontside

mariobehling, 2020-01-16:
The ESP chip is the only module on the backside of the device. Please try to decrease the usage of space on the board and move the chip to the front side.

CloudyPadmal, 2020-01-17:
ESP is about 1/10th of the whole board and even if we managed to bring it to the top, it would not go with the same solder flow as other components as it is a module. Need to evaluate the effort with the outcome we would get from this.

CloudyPadmal, 2020-02-26:
Current version in use is ESP12. Most of the pins in this module are unused in PSLab. With this reasoning, we can replace ESP12 with ESP01 which will enable us to move the component to top.
[image]

CloudyPadmal, 2020-05-18:
Added in #93

PR #113 (2020-05-18): feat: esp01 module
Fixes #107 

- Replaced current ESP12 module with ESP01 to bring it to the top side.
- Routing errors fixed

[image]
<table>
<tr>
<td>[image]</td>
<td>[image]</td>
</tr>
</table>
closed 2020-05-18 by CloudyPadmal