fossasia/pslab-hardware #111: Wave Generator: Triangle wave is deformed

Low012, 2020-01-27:
When I configure the wave generator to output a triangle wave, the wave which is generated is deformed. I checked with the PSLab's oscilloscope and with an external device.

I would expect the wave to look like the wave that is displayed in the app. 

1623Hz triangel wave
[image]

[image]

615 Hz triangle wave
[image]

[image]

justinclift, 2020-01-28:
Ouch. :frowning_face:

19-pragya, 2020-08-15:
can i work on this??

marcnause, 2023-02-06:
I am pretty sure that this is a software issue in the Android app. I will close this issue and add a comment to https://github.com/fossasia/pslab-android/issues/2100 instead.

closed 2023-02-06 by marcnause