fossasia/pslab-hardware #117: Implement feature to turn off PSLab components to save power

mariobehling, 2020-09-15:
The question is if PSLab's components can be selectively turned off to save power? And what's PSLab's rough baseline power consumption? 

Bakcground:
> I'm really interested in developing a project to build OSH wildlife monitoring sensors and it would be amazing if it could be based on, or related to, PSLab. I know many ecologists and conservation biologists who would love access to technology like PSLab to adapt/remix but lack the expertise to do so.
> The key to autonomous sensing, though, is power supply, hence my questions above. Many of PSLab's existing components could be useful in a remote-sensing context, but ideally they should only be powered on when needed. For example, it would be cool if the wireless modules are turned on only when someone wants to establish a data connection in the field. I recall discussing power requirements at a hack-a-thon, and even a Raspberry Pi is way too power-hungry to work well...

penyuan, 2020-09-28:
Thanks for opening this issue!

As someone who has worked in the field, I'd be happy to elaborate more if/when needed.

RafaelLeeImg, 2023-09-20:
The next version v7 will use ARM chip and less power rail to save power.