fossasia/pslab-hardware #136: Populate ICSP header

bessman, 2023-01-28:
### Problem Description

On the latest prototype, the ICSP header (which is used to flash the device with a hardware programmer) is unpopulated. While this header is not commonly used, populating the header makes it easier to e.g. replace the bootloader if necessary.

mariobehling, 2023-09-11:
The ICSP header is populated, but it could be that it was not soldered on the board. We will make a production note that the pins get soldered onto the device as well.

mariobehling, 2023-09-11:
@CloudyPadmal Could you tell us, please why the pin order is reversed relative to the previous prototype? What is the reason for that?

CloudyPadmal, 2023-09-15:
Is it compared to V4 where we have ICSP headers parallel to the board's lengthwise edge?

bessman, 2023-09-15:
Compared to the first V6 prototype. I think it also had the ICSP header perpendicular to the board's lengthwise edge, but the pin order was reversed.

I don't have the boards in front of me right now, and looking through the git history I can't actually find a point where the order was reversed. It's possible I'm mistaken. I will check tonight.

bessman, 2023-09-15:
edabd6f changes the orientation of the header. The pin order is the same then as it is now, so I guess I'm mistaken. Will double check with the actual board later.

bessman, 2023-09-15:
No, on the previous prototype the header is still parallel with the lengthwise edge. I don't know why I thought the order was reversed :thinking: Forget it.

Keeping this issue open for now to make sure we remember to have the header soldered on the production boards.

RafaelLeeImg, 2023-09-20:
Is ICSP headers soldered on PSLab v5? If yes, was it male or female?