fossasia/pslab-hardware #137: Document new buttons and LEDs

bessman, 2023-01-28:
### Problem Description

The new prototype has several new buttons and LEDs. Some are self-explanatory (RESET), while others could use some documentation explaining their purpose.

In particular:

- The BOOT button. Through trial-and-error I found out that keeping BOOT pressed while resetting the device makes it boot into bootloader mode.
- What does the POWER button do?
- What is the purpose of the four white LEDs mentioned in #135?

mariobehling, 2023-09-11:
> The BOOT button. Through trial-and-error I found out that keeping BOOT pressed while resetting the device makes it boot into bootloader mode.

Correct.

> What does the POWER button do?

The user needs to press this button to activate the output of the power converter. It means if the user uses a battery they can use this button to switch on/off the device. Switch on: Press one time. To switch it off press two times within one second to shut down the device.

> What is the purpose of the four white LEDs

With all 4 lid up it means the battery is fully charged. 

To be tested: What happens when connected to USB?

bessman, 2023-09-11:
> To be tested: What happens when connected to USB?

When my prototype is connected to USB, three of the battery LEDs light up and the fourth blinks with a period of ~1 sec.