fossasia/pslab-hardware #139: Power supply not working

bessman, 2023-01-28:
### Problem Description

The power supply pins (PCS, PV1, PV2, PV3) on the new prototype are not working. I recall we changed the chip driving the power supply, correct? Do we need to update the firmware to make the new driver chip work?

CloudyPadmal, 2023-02-11:
New IC: https://www.microchip.com/en-us/product/MCP4822

bessman, 2023-02-20:
The new prototype uses a muxer for the SPI chip select.

### Differences between hardware versions

#### V5

Three SPI chip select outputs:

1. The external CS pin on the SPI header, controlled by C4
2. The gain amplifier for oscilloscope channel 1, controlled by A10
3. The gain amplifier for oscilloscope channer 2, controlled by A7

#### Old prototype

Adds another chip select output for the SD card, controlled by B9.

#### New prototype

Adds another one, for the MCP4822 controlling the power supply. The v5 and the old prototype used a different chip which was connected via I2C.

This brings the total to five, for which we did not have enough free pins available. Thus the muxer.

The muxer is a TI CD74HC4051 Analog Multiplexer and Demultiplexer. It has three inputs and eight outputs. One output at a time can be set high by asserting the three inputs, according to the following table:

| S2 | S1 | S0 | OUT |
|---|---|---|---|
| L | L | L | A0 |
| L | L | H | A1 |
| L | H | L | A2 |
| L | H | H | A3 |
| H | L | L | A4 |
| H | L | H | A5 |
| H | H | L | A6 |
| H | H | H | A7 |

Additionally, all outputs can be set low by setting the E pin high. In the PSLab's case, the E pin is currently connected to ground, so one output will always be high.

The muxer is connected as follows:

Inputs:
A10 (previously CH1 gain) -> S0
A7 (previously CH2 gain) -> S1
B9 (previously SD card) -> S2

Outputs (AX refers to the muxer output pins here):
A0 -> CH1 gain
A1 -> CH2 gain
A2 -> Power supply
A3 -> NC
A4 -> External CS pin
A5 -> SD card
A6 -> NC
A7 -> NC

C4 (which previously controlled the external CS pin) is now NC.

### Way forward

If possible, the firmware should have an abstraction layer around the CS pins, such that setting a certain CS pin can be done in the same way regardless of which underlying pin controls that CS pin. A rudimentary abstraction layer currently exists in pin:manager.h, which maps MCU pin <-> CS pins 1:1. 

Investigate how this abstraction layer could be extended to decouple the CS pins from the MCU pins.