fossasia/pslab-hardware #152: Board Temperature

CloudyPadmal, 2024-06-11:
I just measured board temperature using in-built methods.

| Old Prototype Board - Old Firmware | Old Prototype Board - New Firmware | New Board - New Firmware |
| ---------------------------------- | ---------------------------------- | ----------------------------- |
| [image] | [image] | [image] |

New board MCU gets up-to 43 Celsius when the old prototype MCU is around 30-32 Celsius.

bessman, 2024-06-11:
I don't think this is a problem in and of itself; 43 °C is not worrisome in the sense that it might damage anything. But it does indicate greater power consumption.

This may be related to the resettable fuse problem, and means that battery life in stand-alone mode will be shorter than it could be.

bessman, 2024-06-11:
Any idea what might be causing it?

One thing that's different between the old and new prototype is the addition of the battery indicator LEDs. As noted elsewhere, they are VERY bright. I wonder if they could be the cause of the increases power draw? Would that manifest as greater MCU temperature?

CloudyPadmal, 2024-06-12:
I'm not sure what could be the cause. 

It looks like the RGB LED is mounted in reverse and may be it is drawing current. I plan to look more into this in this week.

CloudyPadmal, 2024-06-14:
I rotated the RGB LED and looks like now it does not heat up anymore.

[image]

However, the RGB LED does not light up. I'll try to solder a new one and see as they are quite plasticky and highly possible to get damaged due to reworking with hot air.

bessman, 2024-06-14:
I suppose this also explains why the RGB LED doesn't work on any of the recent prototypes.

mariobehling, 2024-06-14:
Is this wrong in the schematics?

bessman, 2024-06-14:
Looks correct to me:
[image]
[image]

I wonder how it ended up backwards? :thinking:

cweitat, 2024-06-14:
@bessman are u able to check the LED to see if it has the +- sides as the intended component? because sometimes the chinese made chips have reversed poles.

bessman, 2024-06-14:
Will check tonight unless @CloudyPadmal checks first.

cweitat, 2024-06-14:
One more thing to check. which colour of LED does it happened? White/Green/Blue

CloudyPadmal, 2024-06-14:
This notch in the LED should face towards the MCU. It was like that in the early prototype Rafael sent over to us.

| 3D render | Rafael Prototype |
| --------- | ---------------- |
| [image] | [image] |

But this is not the case with the new boards and somehow the LED is rotated. I just checked the `position` file and the rotation is 0 degrees as expected (D5 - Line 52).

Since this rotation causes the `VCC` and `GND` pins to swap, it might have fried the LED in the first place. Let me double check with a new WS2182B LED.

[image]

bessman, 2024-06-14:
> One more thing to check. which colour of LED does it happened? White/Green/Blue

It's the RGB LED. It can be any color. It should be purple in bootloader, green in application.

> @bessman are u able to check the LED to see if it has the +- sides as the intended component? because sometimes the chinese made chips have reversed poles.

It's the same as in the datasheet:
[image]

Previous prototype (correct orientation):
[image]

Current prototype (wrong orientation):
[image]

Comparison between previous and current prototypes with board traces marked in yellow and LED pins marked in red:
[image]
[image]

CloudyPadmal, 2024-06-14:
Looks like the model got protection against reverse connections.

[image]

cweitat, 2024-06-16:
What can we conclude from here? That the GERBER wasn't followed in regards to the LED direction and because of this, it resulted in faults of ____?

CloudyPadmal, 2024-06-16:
> What can we conclude from here? That the GERBER wasn't followed in regards to the LED direction and because of this, it resulted in faults of ____?

I will finalize this by tomorrow or the day after at the latest.

CloudyPadmal, 2024-06-17:
So I was able to test again.

The RGB LED was burnt and I replaced it with a spare one and now it lights up.

[image]

I did a temperature test, and it seems normal with the new RGB LED.
[image]

Now the conclusion is that, 

- The RGB LED (D5) requires manual de-soldering to correct its position.
- [`PositionFile.csv`](https://github.com/fossasia/pslab-hardware/blob/pslab-v6/docs/position_files/PositionFile.csv) is still correct. I'm not sure why the LED was rotated during assembly.
- I suspect that the fuse was protecting the board from this LED creating a short. Therefore, we will pause the fix we had earlier (shorting the fuse) until a fresh board is tested with the LED in the correct position.

cweitat, 2024-06-18:
I had a chat with the manufacturer side and they want to clarify few items:
[image]
does the yellow circle meant the direction of placement for the component?

More example of the schematics
[image]
[image]
[image]
[image]
[image]

CloudyPadmal, 2024-06-18:
The yellow circle marks the pin 1 in an IC or a component. Also on the silk screen, a special marking shows where a special marking of an IC or a component should align.

[image]

In this figure I have marked such markings for the MCU and the RGB LED.

CloudyPadmal, 2024-06-18:
The standard is that this special marking is next to pin 1. Apparently for WS2812B RGB LED, the special marking is next to pin 3. Maybe this is where it went off-road for the manufacturer.

[image]

CloudyPadmal, 2024-06-21:
## Conclusion

The recently manufactured and assembled boards have an issue.

The issue is that the RGB LED (WS2182B) (D5) is mounted incorrectly, with the VCC (pin 1) and GND (pin 3) swapped. This creates a type of short circuit, rendering the board unusable unless the LED is manually removed and repositioned correctly according to the parameters in the provided PositionFile.csv.

It appears that the manufacturer missed the visual inspection step to verify that the IC pin markings align with the silk screen markings. This issue was caused by the manufacturer’s fault, as all the specifications and files we provided were correct.

@mariobehling @cweitat

cweitat, 2024-06-21:
@CloudyPadmal only Gerber and BOM was given to the manufactuerer. Is the positionfile.csv in the gerber package?

CloudyPadmal, 2024-06-21:
@cweitat `Pick and Place` and `Position` are the same file, with different names.

cweitat, 2024-06-24:
Reply from them: Pick and Place procedure is not able to verify the direction of the components, only the positioning.

CloudyPadmal, 2024-06-24:
Yes, and they got it wrong as it does not require a rotation (0 degrees)

cweitat, 2024-06-24:
They followed the yellow dots as provided in the file as the standard practice as they did with their other customers. And it also impossible to see whether the direction was wrong with just look of it. Moreover, this was only highlighted more than 2 months after it was shipped out

bessman, 2024-06-24:
> They followed the yellow dots as provided in the file as the standard practice

They did not; standard practice is that the yellow dot marks the connection point for pin 1. They have connected pin 3 at the yellow dot.

> it also impossible to see whether the direction was wrong with just look of it.

Incorrect. There are at least two ways to ensure correct orientation with visual inspection:

1. Note the location of the triangular marking next to pin 3, as specified in the datasheet
2. Note that the LED actually turns on when powered (as I believe we instructed them to do?)

cweitat, 2024-06-24:
@bessman Yes. The file given them had the error of  yellow dot placed at pin 3, not pin 1, which is what Padmal brought up earlier messages. So in this case, meaning the Gerber was wrong.

There was no mention of how testing should be done other than LED lighting up. It was raised prior to the production and only the lighting up of LED was mentioned.

For point 2, yes the test was plug in power and see if the LED lights up. No particular instructions was mentioned as to which LED lights up etc. Also, there's no manual for test as well, other than an old one. In this case, the white LED lights up so it's considered as a pass.

CloudyPadmal, 2024-06-24:
I don't know how else to say that it was the manufacturer who messed up the assembly which took us some time to debug 🙆🏻‍♂️

Please don't misinterpret and the production files are still valid and 100% correct.

CloudyPadmal, 2024-06-24:
> And it also impossible to see whether the direction was wrong with just look of it. 

Just looking at it is in fact the visual inspection stage where you identify ICs assembled in wrong orientations. Check my previous comments about silk screen markings. 

This is the standard procedure with PCBWay and JLCPCB.

cweitat, 2024-06-25:
AOI machine file for polarity testing comes from the provided Gerber (which had the dot on pin 3) as well. So the test would have passed any ways.

CloudyPadmal, 2024-06-25:
Dot is in pin 1 as mentioned earlier. 

I'm repeating the same thing and I don't think this discussion yields anything.

cweitat, 2024-06-26:
@CloudyPadmal yes I understand that as well. Therefore I left the communication to the manufacturer with Mario now.

CloudyPadmal, 2024-06-26:
Sounds good. I'm closing this issue as now we know that the manufacturer did not follow the design files correctly and we will manually fix the LEDs with their supposed orientation before shipping out to customers.

closed 2024-06-26 by CloudyPadmal