fossasia/pslab-hardware #154: Missing release of the final v6 as a tagged stable release

froscon2024, 2025-03-04:
The current releases are pre-releases https://github.com/fossasia/pslab-hardware/releases
A stable, final release of the v6 that is in production since over a year would be great for archive and documentation purpose.

bessman, 2025-03-04:
@CloudyPadmal @mariobehling Could you confirm which commit was used for production so we can tag it?

froscon2024, 2025-03-04:
The date code printed on the v6 i have is: 25.12.2023
It was missing the ESP-01 header i could see on some renderings of the v6. I soldered the header in myself to be able to connect and remove the ESP-01 whenever i want.

When i take a look at https://github.com/fossasia/pslab-hardware/commits/pslab-v6/
then i can see that the only two changes that have been merged after 25.12.2023.

The PCB i have is already with the silk screen updates: https://github.com/fossasia/pslab-hardware/pull/150/files
And those got merged January 2024 but the PR is from end of 2023 and looks to be done for production but merged into this repo after producing the PCB's with the PR included.

So i think the production code of the v6 is the code that is recent this days in this repo.
I think when you now make a stable release with the current code, this would be exactly the production version of the v6. You can then remove the older pre-release parts and then there would be just stable v5 and stable v6 release in this repo.